betaflight/unified-targets #1365: no gyro problem  choosing right firmware

sarojlawot, 2025-05-08:
###

# # Problem description

# # no gyro

# ###

# version
# Betaflight / AT32F435M (A435) 4.5.2 Mar 22 2025 / 13:35:42 (024f8e13d) MSP API: 1.46
# config rev: 3eedb40
# board: manufacturer_id: TEBS, board_name: TBS_LUCID_FC

# status
MCU AT32F435 Clock=288MHz, Vref=3.26V, Core temp=51degC
Stack size: 2048, Stack address: 0x2002fff0
Configuration: UNCONFIGURED, size: 4102, max available: 16384
Devices detected: SPI:0, I2C:1
Gyros detected: locked
, BARO=BMP388
OSD: MSP (53 x 20)
BUILD KEY: f0a41cf1b5166e1d3e28aa6388b2b64b (4.5.2)
System Uptime: 35 seconds, Current Time: 2025-05-08T10:18:25.127+00:00
CPU:0%, cycle time: 0, GYRO rate: 0, RX rate: 0, System rate: 9
Voltage: 4 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
FLASH: JEDEC ID=0x00ef4017 8M
GPS: NOT ENABLED
Arming disable flags: NOGYRO BOOTGRACE CLI MSP

# flash_info
Flash sectors=128, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=8388608 JEDEC ID=0x00ef4017
Partitions:
  0: FLASHFS   0 127
FlashFS size=8388608, usedSize=8388608

# dump master

# version
# Betaflight / AT32F435M (A435) 4.5.2 Mar 22 2025 / 13:35:42 (024f8e13d) MSP API: 1.46
# config rev: 3eedb40

# start the command batch
batch start

board_name TBS_LUCID_FC
manufacturer_id TEBS

# name: -

# resources
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 B05
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 A00
resource MOTOR 6 A01
resource MOTOR 7 A02
resource MOTOR 8 A03
resource SERVO 1 B07
resource SERVO 2 B08
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
[... 1103 lines cut ...]
# timer show

Currently active Timers:
-----------------------
TIM1:
    CH1 : LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4:
    CH2 : SERVO 1
    CH3 : SERVO 2
TIM5: FREE
TIM8:
    CH2 : DSHOT_BITBANG 3
    CH3 : DSHOT_BITBANG 2
TIM9:
    CH2 : CAMERA_CONTROL
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM20: FREE

#

haslinghuis, 2025-05-08:
Please flash the updated target for your board:

Gyro | Target
--- | ---
MPU6000 | TBS_LUCID_PRO_FC
ICM42688P | TBS_LUCID_FC

sarojlawot, 2025-05-09:
choose it right firmware now 
moter is not beeping
fc is tbs_lucid_pro_fc
gyro is showen but moter is not detected